antmicro/rowhammer-tester #2: Set up Sphinx and readthedocs documentation generation for this repository

mithro, 2020-10-15:
You colleague, @rw1nkler can provide a bunch of advice here.

mithro, 2020-10-21:
FYI -- @kgugala can you give @rw1nkler some time to help get this set up for this team?

rw1nkler, 2020-10-21:
@mithro I already talked with the team about this project and gave them some advice. 
Hopefully, the documentation will show up soon. 

FYI @jedrzejboczar

kgugala, 2020-10-21:
Initial (autogenerated) documentation is already there - https://antmicro.github.io/litex-rowhammer-tester/. Need to update it with some nicer theme and additional information. Currently it just lists registers in the system.

tjurtsch, 2021-02-26:
Sphinx documentation on RTD is here https://litex-rowhammer-tester.readthedocs.io/en/latest/index.html

closed 2021-02-26 by tjurtsch