antmicro/rowhammer-tester #4: Why is EtherBone bridge so slow?

mithro, 2020-10-21:
> Thanks Jedrzej. One question -- why is EtherBone so slow? 1MB/22s = 45KB/s. I was expecting a bandwidth that's two orders of magnitude larger. If reading out the module takes an hour, that's going to really slow down experiments. Will this get faster for future implementations and/or for higher-speed FPGA boards?

_Originally posted by @yoongu in https://github.com/antmicro/litex-rowhammer-tester/issues/3#issuecomment-713639055_

mithro, 2020-10-21:
First question @jedrzejboczar -- Are you using the bridge over the uart or 100Mbit Ethernet on the Arty board?

mithro, 2020-10-21:
> If reading out the module takes an hour, that's going to really slow down experiments. Will this get faster for future implementations and/or for higher-speed FPGA boards?

To answer this part question, see issue #5. Reading the memory regions over the wishbone and checking on the host is a temporarily solution, the plan has always been to do this on the FPGA using the BIST functionality at the full speed of the DDR memory controller.

That said, the wishbone performance is still slower than expected.

mithro, 2020-10-21:
The wishbone bridge over 100Mbit Ethernet should be able to get speeds up to ~80 Mbit (10 Megabyte/second) if used correctly.

As the wishbone bridge has mainly be used for debugging and simple control of CSR registers, not much time has been taken to look into bulk transfer performance. The actual Etherbone protocol the bridge is based on **was** designed to be reasonably performant for bulk transfers, so fixing this should be pretty easy.

A couple of things that make sense to check;
 - Make sure you are using the bridge other Ethernet rather than over UART which has a maximum of ~100 kilobyte per second.
 - The default configuration of the bridge reads only 4 bytes per transaction. When doing bulk transfers, you really want to read or write ~(1500 bytes - (byte of packet overhead) - (bytes of wishbone instruction overhead)) which should be at least >1000 bytes. (This may require a few small fixes to the Etherbone peripherals).
 - For absolutely best performance you probably should pipeline / queue the read requests (but probably not worth doing given how low latency directly connected Ethernet should be).
 - On 1G interface you might need to enable jumbo frames to get the higher bandwidth than >100 megabits.

yoongu, 2020-10-21:
BIST works best when data-patterns have small entropy and the number of errors is small, both of which are true in most experiments. But I can also easily see cases where they're not. For example, playing retention roulette: turning off refreshes for a full minute and reading back all the data. This'll yield patches of 0s and 1s reflecting the native polarity of the individual cells.

While ~10MB/s sure sounds nice (transferring a full bank would happen in seconds), if that's too much effort then ~1MB/s would be tolerable in the mid-term. From what @mithro said, it seems like there are some low hanging fruit that might get us there.

mithro, 2020-10-21:
@yoongu 1 Megabyte per second is only ~10 Megabits, that should be *easily* doable. 100 Megabytes per second is ~100 Megabits a second, so probably not going to be reached on the Arty (which only has 100M Ethernet) but it should **easily** be doable on boards which have 1G Ethernet, I would expect we should be able to get more towards the 400-500 Megabyte per second on these boards.

yoongu, 2020-10-21:
Fantastic. Getting to within an order of magnitude of the line-rate is a win in our book.

jedrzejboczar, 2020-10-22:
@mithro Yes, I was using the Ethernet bridge. I was transmitting/receiving in 16-word (64 byte) bursts. I was able to double the write performance (from ~1MBps to ~1.9MBps) using 1020 byte bursts, which was the maximum size. However it does not affect read  the performance (still ~50KBps).

xobs, 2020-10-24:
I don't have the ability to test etherbone, but I'll be adding streaming support to the wishbone-tool USB bridge. If someone wants to test ethernet support, I can add it there, too.

Whats the biggest packet that etherbone supports?

jedrzejboczar, 2020-10-26:
@xobs For the standard 1500 byte packets it should be 1500-66=1434 (see [the document](https://journals.aps.org/prab/pdf/10.1103/PhysRevSTAB.15.082801), section IX.A.1). With jumbo frames this could probably reach up to 9000-66=8934.

jedrzejboczar, 2021-05-13:
With the EtherBone improvements in Litex upstream and increased burst sizes on our side we now have decent performance, and when we need to fill/scan the whole DRAM we use the BIST modules which offer superior speeds, so closing this issue for now.

closed 2021-05-13 by jedrzejboczar