antmicro/rowhammer-tester #7: Payload Executor output doesn't match the written pattern

sqazi, 2020-10-29:
$ git log --oneline |head -4
b76dd14 Change IP address to match setup
9e35e50 Resuse address bits in NOOP instructions for TIMESLICE
36dd7ec Add support for multiple ranks
4ea1ee2 Add payload execution scratchpad memory for data from READ commands

$ make upload
./third_party/xc3sprog/xc3sprog -c nexys4 build/arty/gateware/arty.bit
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Using Libftdi, 

(venv) sqazi@sqazi-glaptop:~/fpga/litex-rowhammer-tester/scripts$ python mem.py
SDRAM initialization:
control_cmds: [0, 1]
Release reset (ctrl)
Bring CKE high (ctrl)
Load Mode Register 2, CWL=5 (cmd)
Load Mode Register 3 (cmd)
Load Mode Register 1 (cmd)
Load Mode Register 0, CL=6, BL=8 (cmd)
ZQ Calibration (cmd)

Read leveling:
Module 0
Bitslip 00: |11111111111100000000000000000000|
Bitslip 01: |00000000000000111111111111111000|
Bitslip 02: |00000000000000000000000000000011|
Bitslip 03: |00000000000000000000000000000000|
Bitslip 04: |00000000000000000000000000000000|
Bitslip 05: |00000000000000000000000000000000|
Bitslip 06: |00000000000000000000000000000000|
Bitslip 07: |00000000000000000000000000000000|
Best: bitslip = 1, delay = 21 (+-7)
Module 1
Bitslip 00: |11111111111100000000000000000000|
Bitslip 01: |00000000000000111111111111110000|
Bitslip 02: |00000000000000000000000000000011|
Bitslip 03: |00000000000000000000000000000000|
Bitslip 04: |00000000000000000000000000000000|
Bitslip 05: |00000000000000000000000000000000|
Bitslip 06: |00000000000000000000000000000000|
Bitslip 07: |00000000000000000000000000000000|
Best: bitslip = 1, delay = 21 (+-7)

Memtest (basic):
1. ['0x55555555', '0x55555555', '0xaaaaaaaa', '0xaaaaaaaa']
2. ['0xaaaaaaaa', '0xaaaaaaaa', '0x55555555', '0x55555555']
Mem ok!

Memtest (random):
OK

(venv) sqazi@sqazi-glaptop:~/fpga/litex-rowhammer-tester/scripts$ python execute_payload.py 
Transferring the payload ...
Executing ...
Finished
Scratchpad contents:
0x00000000:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000010:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000020:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000030:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000040:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000050:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00000060:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
[... 482 lines cut ...]
0x00001e90:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ea0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001eb0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ec0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ed0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ee0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ef0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f00:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f10:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f20:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f30:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f40:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f50:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f60:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f70:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f80:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001f90:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001fa0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001fb0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001fc0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001fd0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001fe0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............
0x00001ff0:  f7 ff fd 7a ff ef f5 9f f5 9f 00 00 00 00 00 00  ...z............

Let me know if you need any additional information / experiments.

jedrzejboczar, 2020-10-29:
Thank you, this should be enough for now. I'll have to do some tests on my side and see why the data is incorrect.

jedrzejboczar, 2020-11-04:
@sqazi I found the issue and fixed it.

There was one problem in both simulation and hardware with EtherBone data conversion and another problem in hardware only. The problem in hardware was that all READ commands have to be sent on `rdphase` that is expected by the PHY in order to have read data aligned correctly. For the current setup `rdphase=2`  but it depends on the frequency of operation. Payload Executor will now send READ commands on correct phase and all other commands on phase 0. When adding the status CSRs with timing/geomertry information we will also include `rdphase`.

I merged the changes to master and added option to run Row Hammer using Payload Executor with the following results:
```
» python rowhammer.py --nrows 512 --read_count 10e6 --pattern 01_in_row --row-pairs const --const-rows-pair 54 133 --plot --payload-executor

Preparing ...

Filling memory with data ...
................................
Running row hammer attacks ...
  Payload size =  0.12KB /  4.00KB
  Payload row toggle count = 10.49M

Transferring the payload ...

Executing ...

Verifying attacked memory ...
................................
row_errors for row=   53: 50
row_errors for row=   55: 88
```
Note that with `--payload-executor` only the first row of the two provided is hammered, so row 54 in this case.

And just for reference these are payload execution results on Arty:
```
» python execute_payload.py
Transferring the payload ...

Executing ...
Finished

Scratchpad contents:
0x00000000:  ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff  ................
0x00000010:  ff ff ff ff ff f7 ff ff ff ff ff ff ff ff ff ff  ................
0x00000020:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000030:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000040:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000050:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000060:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000070:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000080:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x00000090:  00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff  .."3DUfw........
0x000000a0:  33 44 55 66 77 88 99 aa bb cc dd ee ff 00 11 22  3DUfw.........."
0x000000b0:  66 77 88 99 aa bb cc dd ee ff 00 11 22 33 44 55  fw.........."3DU
0x000000c0:  99 aa bb cc dd ee ff 00 11 22 33 44 55 66 77 88  ........."3DUfw.
0x000000d0:  cc dd ee ff 00 11 22 33 44 55 66 77 88 99 aa bb  ......"3DUfw....
0x000000e0:  ff 00 11 22 33 44 55 66 77 88 99 aa bb cc dd ee  ..."3DUfw.......
0x000000f0:  22 33 44 55 66 77 88 99 aa bb cc dd ee ff 00 11  "3DUfw..........
0x00000100:  55 66 77 88 99 aa bb cc dd ee ff 00 11 22 33 44  Ufw.........."3D
0x00000110:  88 99 aa bb cc dd ee ff 00 11 22 33 44 55 66 77  .........."3DUfw
0x00000120:  cc dd ee ff 00 11 22 33 44 55 66 77 88 99 aa bb  ......"3DUfw....
0x00000130:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000140:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000150:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000160:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000170:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000180:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000190:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001a0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001b0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001c0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001d0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001e0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000001f0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```
The data for `0x00000000` and `0x00000010` may vary as this is the memory that has not been filled with any pattern.

closed 2020-11-04 by jedrzejboczar

sqazi, 2020-11-04:
Thank you!  The output I see matches yours.