antmicro/rowhammer-tester #8: Add more packages to setup documentation

yoongu, 2020-11-01:
I'm on Debian so there might be some differences, but I had to install two more packages. They apparently weren't dependencies of the ones already listed.

https://packages.ubuntu.com/bionic/autoconf
https://packages.ubuntu.com/bionic/python3-wheel

yoongu, 2020-11-01:
Another Debian workaround, this time while installing Vivado. The Java GUI has a broken dependency and we have to force it to run in the CLI.

https://bbs.archlinux.org/viewtopic.php?pid=1910277#p1910277

yoongu, 2020-11-01:
Another Debian workaround, this time while running Vivado. This is all for Buster by the way (10.4).

```
couldn't load file "librdi_commontasks.so": libtinfo.so.5: cannot open shared object file
```

```
$ sudo apt install libtinfo-dev
$ sudo ln -s /lib/x86_64-linux-gnu/libtinfo.so.{6,5}
```

yoongu, 2020-11-02:
At this point, I'm pretty sure that the only micro USB cable I have at home is charge-only. Can't get Arty should show up on `lsusb` or the like. The kernel modules are definitely loaded for ftdi_sio and usbserial. Guess I'll have to make a quick trip to MicroCenter.

yoongu, 2020-11-05:
Another Debian workaround, the ```ftdi_sio``` module isn't loaded by default.

```
sudo modprobe ftdi_sio
cat /proc/modules | grep ftdi_sio
```

jedrzejboczar, 2021-05-13:
The dependencies already have been added to the documentation https://litex-rowhammer-tester.readthedocs.io/en/latest/general.html#installing-dependencies, so closing this one. The other workarounds are probably too specific to include them in the general docs.

closed 2021-05-13 by jedrzejboczar