antmicro/rowhammer-tester #11: Read IP address as an environment variable.

yoongu, 2020-11-05:
Currently it's hardwired in various files (e.g., Makefile, arty.py) which leads to merge conflicts.

jedrzejboczar, 2020-11-06:
It is actually already possible if I understand you correctly, e.g.
```
» make 2>1 | head
python gateware/arty.py --ip-address 192.168.100.50 --mac-address 0x10e2d5000001 --udp-port 1234
» IP_ADDRESS=192.168.100.60 make 2>1 | head
python gateware/arty.py --ip-address 192.168.100.60 --mac-address 0x10e2d5000001 --udp-port 1234
```
Makefile [has variables](https://github.com/antmicro/litex-rowhammer-tester/blob/master/Makefile#L6-L9) that are then passed as command line arguments to `arty.py`. So doing `export IP_ADDRESS=192.168.100.60` should work. Or do you mean removing command line arguments from `arty.py` and using only environmental variables?

jedrzejboczar, 2020-11-06:
I just noticed `IP_ADDRESS` was not being used in the Makefile `srv` target, so I fixed that.

closed 2020-11-12 by jedrzejboczar