antmicro/rowhammer-tester #21: Row addresses seem non-linear when hammering

sqazi, 2020-12-09:
This is not necessarily a problem to be solved by antmicro, but please double check addressing.  We will also do some experiments on our end to see if there is a pattern to how the hammering flips bits.

We would expect to see up to 4 rows with flips for a single pair, usually near the rows accessed. 

This is an example of what we actually see instead:

$ python rowhammer.py --nrows 512 --read_count 10e6 --pattern 01_in_row --row-pairs const --const-rows-pair 10 12 --no-refresh
Using generated target files in: ../../build/zcu104

Preparing ...

Filling memory with data ...
................................
Disabling refresh ...

Running row hammer attacks ...
  Iter 0 / 1 Rows = (   10,   12), Count = 10.25M / 10.00M  

Reenabling refresh ...

Verifying attacked memory ...
................................
row_errors for row=   36: 315
row_errors for row=   44: 505
row_errors for row=   52: 239

==================================

The addresses seem consistent across runs, so either the DRAM itself has some fancy mapping or there is a problem with partitioning of the address bits into its components.

sqazi, 2020-12-09:
Actually, I think the problem is pretty obvious if we run a few tests:

pair      flips
0 1        8
0 2        4  12
0 3        4   8   16
0 4        4  12   20

The rows numbers printed out for errors are 4x the actual row number.  For example, if we hammer 0 and 1, we expect 2 to flip.  2*4 = 8.  Same with the other examples here.

jedrzejboczar, 2020-12-10:
Yes, thanks for reporting this, there was indeed an issue with address encoding/decoding. I posted a fix in https://github.com/antmicro/litex-rowhammer-tester/commit/cf31b52734ee0fe0859ee5e4925006d8bdf02118. Please confirm that it's ok now. Now I get results suggesting that the rows grow linearly.

Also, you should be able to use the `hw_rowhammer.py` script with the same arguments to perform the test using BIST modules. It should be much faster and test whole memory range.

sqazi, 2020-12-11:
It works now.   Thank you.

closed 2020-12-14 by jedrzejboczar