antmicro/rowhammer-tester #22: Single read on ZCU104 over PS Ethernet may result in multiple CSR read accesses.

jedrzejboczar, 2020-12-16:
I observed that doing memory initialization on ZCU104 over the Ethernet, the data read from the VexRiscV CPU's UART (CPU in PL) is malformed. This looks like part of the UART bytes are lost.

I think that this issue is due to the fact that a single memory access from PS can result in multiple wishbone reads in PL.

The current way the access is performed is as described here https://github.com/antmicro/litex-rowhammer-tester/blob/7ceb33f3cc57611fc3711546af1f425f80e7f955/rowhammer_tester/targets/zcu104.py#L177. The problem most likely arises in this data width converter `AXILite(128) -> AXILite(32)`. A single access results in 4 32-bit accesses. Due to the way LiteX CSR address decoders work it may be that all these accesses go to the same CSR. UART read CSR works like a FIFO, and an access consumes the current byte, which would explain why bytes are lost.

If this is the case, than it will cause problems with using the [reader_error_offset CSR](https://antmicro.github.io/litex-rowhammer-tester/reader.html#reader-error-offset) which also [works as a FIFO](https://github.com/antmicro/litex-rowhammer-tester/blob/7ceb33f3cc57611fc3711546af1f425f80e7f955/rowhammer_tester/gateware/bist.py#L263).

jedrzejboczar, 2021-02-18:
Thanks for this fix. Unfortunatelly with this change I still get the issues with UART corruption. We may need to capture the AXI read signals and then check the whole access path in a simulation to better see what is causing the issue.

I only cherry-picked the commit you mentioned, because when I tried to move to the newest version of litex libraries I had problems with litex_server - first it was failing on the new `probe()` method (I guess we do not handle this in our etherbone server code) and even after removing that part I still had timeouts. So proper bump of litex dependecy will yet require some changes in our code.

enjoy-digital, 2021-02-18:
@jedrzejboczar: OK thanks for the test. Capturing the Wishbone transactions over UARTBone with LiteScope would indeed be useful to understand.

closed 2021-03-01 by jedrzejboczar