antmicro/rowhammer-tester #25: hw_rowhammer.py only supports single pattern

sqazi, 2021-01-06:
The original slower implementation supports a variety of patterns, but the script for the hardware implementation only supports single pattern used across all the rows.  At the very least, we think it should support patterns that fill the hammered rows with one pattern and the remaining rows with its inverse.  In our experience elsewhere, this helps with the probability of getting bit flips.

In our software tool (https://github.com/google/hammer-kit), the way we specify data patterns might interest you.  We specify a divisor and we divide the row number by this divisor and look at the remainder.  We check the remainder against a bitmask, and if the corresponding bit is set, then we consider it a "victim row".  Victim rows get the inverse pattern from all other rows.

sqazi, 2021-01-12:
Is there any progress on this?

jedrzejboczar, 2021-01-12:
I've been working on this, it is a bit more problematic than I initially thought.
I pushed a WIP version to https://github.com/antmicro/litex-rowhammer-tester/tree/jboc/dev.

If I'm correct the algorithm used in hammer-kit is something like the following:
```
# as in: https://github.com/google/hammer-kit/blob/40f3988cac39e20ed0294d20bc886e17376ef47b/hammer.c#L270
def test_matching(mod, nrows, bitmask):
    nbits = bits_for(nrows)
    print(f'# nrows = {nrows:3} mod = {mod:3} bitmask = {bitmask:0{nbits}b}')
    for row in range(nrows):
        rowmod = row % mod
        rowbit = 1 << rowmod
        matcher = bitmask & rowbit
        matchstr = (' <' if matcher != 0 else '')
        print(f'row = {row:3} = {row:0{nbits}b}  %{mod} = {rowmod:0{nbits}b}  bitmatch={rowbit:0{mod}b}' + matchstr)
```
The operation `rowbit = 1 << rowmod` is quite expensive to do in FPGA, but should be ok for low numbers of rows (as far as I understand `mod` in hammer-kit can be at most 64 due to 64bit integer being used, so we can use at most 6 row address bits).

I've added an initial test of the functionality for BISTWriter: 
https://github.com/antmicro/litex-rowhammer-tester/blob/56fa62aa62c0f3d7da57b9a839eacd8f9cab88cd/tests/test_bist.py#L182-L221
Can you please take a look at the `modmask` and `bitmask` used there and say if the following result (list of commands: `(address, iswrite, data)`) is the expected output?
```
test_row_data_invertion (tests.test_bist.TestWriter) ...
0x00000000 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000001 1 0x55555555555555555555555555555555
0x00000002 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000003 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000004 1 0x55555555555555555555555555555555
0x00000005 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000006 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000007 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000008 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x00000009 1 0x55555555555555555555555555555555
0x0000000a 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x0000000b 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x0000000c 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x0000000d 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x0000000e 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
0x0000000f 1 0xaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
```

sqazi, 2021-01-12:
Thanks for doing this!

Your understanding of the algorithm in hammer-kit is correct.  Yes, mod is usually a small number.  The code supports up to 64, but in practice, we don't get anywhere near that.  It's usually a small number under 8.  I would be more than happy with an upper bound of 32.

The output matches what I would expect from the code.  I don't understand the iswrite bit, but the address and corresponding data is what I would expect.

sqazi, 2021-01-12:
Just so that I understand correctly, the modmask is the divisor (power of 2) minus 1, right?

jedrzejboczar, 2021-01-13:
Yes, you're right. It's just doing modulo by masking. The iswrite bit is not relevant, I just printed it due to the way I was testing this. Good to here the algorithm is ok, I'll continue to work on it in this shape then.

closed 2021-01-18 by jedrzejboczar