antmicro/rowhammer-tester #27: More of a question than a bug: can scripts programmatically obtain memory parameters?

sqazi, 2021-01-07:
Motivation: we want to be able to compile valid payloads for whatever part that is plugged into the board at the given moment from part-agnostic descriptions.  So, it would be nice to be able to look up timing parameters for the part (e.g. tREFI, tFAW) from the scripts.  Is this possible?

jedrzejboczar, 2021-01-07:
During build a file `litedram_settings.json` is generated in target's build directory (e.g. `build/zcu104/litedram_settings.json`). It can be loaded in the scripts using provided [helper function](https://github.com/antmicro/litex-rowhammer-tester/blob/master/rowhammer_tester/scripts/utils.py#L73). Among other controller settings it also contains timing parameters, however these parameters are in units of memory controller cycles. If values in nanoseconds are needed than we would have to expose them in a similar manner.

Example contents of `litedram_settings.json`:
```json
{
    "cmd_buffer_depth": 8,
    "cmd_buffer_buffered": false,
    "read_time": 32,
    "write_time": 16,
    "with_bandwidth": false,
    "with_refresh": 1,
    "refresh_zqcs_freq": 1.0,
    "refresh_postponing": 1,
    "with_auto_precharge": true,
    "address_mapping": "ROW_BANK_COL",
    "phy": {
        "phytype": "USPDDRPHY",
        "memtype": "DDR4",
        "databits": 64,
        "dfi_databits": 128,
        "nphases": 4,
        "rdphase": 2,
        "wrphase": 2,
        "cl": 9,
        "read_latency": 8,
        "write_latency": 3,
        "nranks": 1,
        "cwl": 9,
        "cmd_latency": 1,
        "cmd_delay": null,
        "is_rdimm": false
    },
    "geom": {
        "bankbits": 3,
        "rowbits": 16,
        "colbits": 10,
        "addressbits": 16
    },
    "timing": {
        "tRP": 3,
        "tRCD": 3,
        "tWR": 3,
        "tWTR": 2,
        "tREFI": 977,
        "tRFC": 45,
        "tFAW": 5,
        "tCCD": 1,
        "tRRD": 2,
        "tRC": 7,
        "tRAS": 5,
        "tZQCS": 32,
        "fine_refresh_mode": "1x"
    }
}
```

jedrzejboczar, 2021-01-07:
While checking https://github.com/antmicro/litex-rowhammer-tester/pull/26 I noticed that the `payload_executor_attack` function could use timings from `litedram_settings.json` so I fixed that. It now loads the settings [here](https://github.com/antmicro/litex-rowhammer-tester/blob/master/rowhammer_tester/scripts/rowhammer.py#L254) and then [uses the timings](https://github.com/antmicro/litex-rowhammer-tester/blob/master/rowhammer_tester/scripts/rowhammer.py#L167-L168), so it can serve as a reference if the timings in units of "memory controller clock cycles" are sufficient.

closed 2021-01-13 by jedrzejboczar