antmicro/rowhammer-tester #42: Supported boards

pjattke, 2021-01-30:
Dear developers,

in your README you mention _Currently, only the Arty-A7 (xc7a35t) FPGA board is supported (TARGET=arty)_ but at the same time also have a section about the _ZCU104 board_. Can I assume that this is a mistake and both boards are indeed supported? If so, could you please describe if there are any differences and which one would you recommend (e.g., is better tested)?

Thank you very much!

kgugala, 2021-01-30:
Hi @pjattke indeed this is a mistake, we support both - Arty and ZCU104. I'd say both of them are equally tested. Keep in mind that Arty is targeting DDR3, while ZCU DDR4 (SODIMM modules).

Thanks for reporting this - we'll update the README.

BTW We're also working on a HW for testing LDPPR4 (see https://github.com/antmicro/lpddr4-test-board and https://github.com/antmicro/lpddr4-testbed) and tester system support for it

pjattke, 2021-01-31:
@kgugala Great, thanks for the explanation! I have clarified this in the PR.

closed 2021-01-31 by pjattke