antmicro/rowhammer-tester #43: Questions about application of LiteX

pjattke, 2021-01-30:
Dear developers,

I'm totally new in this hardware world, so sorry if these questions might look obvious to you. I'm wondering how exactly we can use this DRAM controller implementation in our research. From the diagram in the README I can see that LiteX is running on a FPGA which is connected to a host (via Ethernet) and via PHY to DRAM. 

Does PHY mean that the DRAM is plugged into the FPGA, e.g., by adding a DIMM socket to the FPGA where the DRAM can be plugged into? Because I saw that the Arty A7 only support DDR3L (not the regular DDR3 for PC that we want to focus on). Or is the DRAM only simulated by LiteX based on the given properties, e.g., the DDR standard?

Thanks a lot for your explanation!

kgugala, 2021-01-30:
Hi @pjattke on Arty target there is no easy option to change the DRAM (you'd have to replace on-board's chip). Arty target is more like "my first row hammer board" target ;). It is a cheap devboard to enable developers to start playing with the system.

On ZCU104 you can change the SODIMM memories and test them.

We have also designed a dedicated HW https://github.com/antmicro/lpddr4-test-board https://github.com/antmicro/lpddr4-testbed to be able to test different memories. The design is open source so anybody can adapt it for their needs. We're currently working on the software support for it.

pjattke, 2021-01-31:
Ah, I see! Thank you, @kgugala, for the information.

closed 2021-01-31 by pjattke