antmicro/rowhammer-tester #46: Using BIST on a single row

sqazi, 2021-02-05:
How do I compute the offset to pass into hw_memtest if I want to just find errors in a single row?

jedrzejboczar, 2021-02-05:
The offset in `hw_memset`/`hw_memtest` is a regular bus address, but address 0 corresponds to the beginning of DRAM address space. `size` is also a number of bytes. The functions then calculate values to be sent to the BIST modules. So in order to get correct offset you have to subtract `wb.mems.main_ram.base` from a regular bus address. You can use `DRAMAddressConverter.encode_bus` to get correct addresses. 

For example, to test row 10, all banks you could do:
```python
row = 10

conv = DRAMAddressConverter.load()
start = conv.encode_bus(bank=0, row=row, col=0)
end = conv.encode_bus(bank=0, row=row + 1, col=0)

offset = start - wb.mems.main_ram.base
hw_memtest(wb, offset, end - start, [0xaabbccdd])
```
Note that it takes advantage of the fact we use ROW-BANK-COL (MSB to LSB) address encoding. 

If you instead want to only test a single bank, then you could instead do:
```python
bank = 3
row = 10

conv = DRAMAddressConverter.load()
start = conv.encode_bus(bank=bank, row=row, col=0)
end = conv.encode_bus(bank=bank + 1, row=row, col=0)

offset = start - wb.mems.main_ram.base
hw_memtest(wb, offset, end - start, [0xaabbccdd])
```
However for the edge case when the bank is the last one (e.g. there are 8 banks and we use `bank = 7`) then you would need to use:
```python
start = conv.encode_bus(bank=7, row=row, col=0)
end = conv.encode_bus(bank=0, row=row + 1, col=0)
```
so in a more generic solution we need to do the wrapping:
```python
def hw_memtest_bank(row, bank):
    settings = get_litedram_settings()
    nbanks = 2**settings.geom.bankbits

    conv = DRAMAddressConverter.load()
    start = conv.encode_bus(bank=bank, row=row, col=0)
    end = conv.encode_bus(bank=(bank + 1) % nbanks, row=row + (bank + 1)//nbanks, col=0)

    offset = start - wb.mems.main_ram.base
    hw_memset(wb, offset, end - start, [0xaabbccdd])
```

Maybe the interface of `hw_memset`/`hw_memtest` is a little inconsistent in that it uses regular bus address/size but it still requires to subtract the main_ram base address. We could probably do the subtraction inside the functions so that bus addresses could be used directly (and just have an assertion that checks that the address actually maps to DRAM memory). I know think that would make more sense, but it also requires changing the usage in all the files. If you think it is worth modifying we could probably make the change in the interface.

jedrzejboczar, 2021-02-08:
I think we can close it now.

closed 2021-02-08 by jedrzejboczar