antmicro/rowhammer-tester #49: Can't generate First Stage Boot Loader (FSBL) for ZCU104

rolin-ron, 2021-02-10:
Was trying to follow the instructions for [ZCU104 boot image preparation](https://github.com/antmicro/litex-rowhammer-tester/blob/master/firmware/zcu104/README.md). The instructions to generate the FSBL are off of Xilinx's site:

- https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842462/Build+PMU+Firmware
- https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841798/Build+FSBL

However, what is needed is the hardware description file (HDF) which gets generated from Vivado but I get this error:
```
Vivado% write_hwdef -force -file filename.hdf
INFO: [Vivado_Tcl 4-424] Cannot write hardware definition file as there are no IPI block design hardware handoff files present
```
It looks like from the generated verilog file, the ARM core is directly instantiated instead of the using the Xilinx Block Design. Here is snipped of code from the generated zcu104.v (line 25385 also attached file)
[zcu104.v.zip](https://github.com/antmicro/litex-rowhammer-tester/files/5961445/zcu104.v.zip)

```
PS8 PS8(
	.MAXIGP0ARREADY(main_axi_ps_ar_ready),
	.MAXIGP0AWREADY(main_axi_ps_aw_ready),
	.MAXIGP0BID(main_axi_ps_b_payload_id),
	.MAXIGP0BRESP(main_axi_ps_b_payload_resp),
	.MAXIGP0BVALID(main_axi_ps_b_valid),
	.MAXIGP0RDATA(main_axi_ps_r_payload_data),
```

A few questions:

1. What version of Vivado were you using to generate the FSBL?
2. How did you generate the FSBL?
3. Do you have prebuilt binaries for ZCU104?

jedrzejboczar, 2021-02-11:
Hi. Regarding 3) I published the prebuilt binaries as a release https://github.com/antmicro/litex-rowhammer-tester/releases/tag/zcu104-v0.1. It should be the most straightforward way to set up the board. We have future plan to move to u-boot SPL so that the build process could be fully automated.

For some more details: to generate FSBL I used Vivado v2018.3, I created a minimal Vivado project with instantiation of Zynq UltraScale+ MPSoC with a minimal wrapper and used it to generate the required files, this an example script run with `hsi -mode batch -source cmds.tcl`:
```tcl
#!/usr/bin/tclsh
set hwdsgn  [open_hw_design /path/to/vivado/project_1/project_1.sdk/design_1_wrapper.hdf]
generate_app -hw $hwdsgn -os standalone -proc psu_cortexa53_0 -app zynqmp_fsbl -compile -sw fsbl -dir ./fsbl/
generate_app -hw $hwdsgn -os standalone -proc psu_pmu_0 -app zynqmp_pmufw -compile -sw pmufw -dir ./pmufw
```

rolin-ron, 2021-02-11:
Awesome! Thanks so much! I will just use the compiled binaries.

Good call on the u-boot SPL as Xilinx no longer supports HDF. Did you want to leave the issue open until you move to u-boot SPL or just close it?

jedrzejboczar, 2021-02-11:
I think we can close this issue for now. I added a link to the [README](https://github.com/antmicro/litex-rowhammer-tester/blob/master/firmware/zcu104/README.md) so it should now point new users to the releases page. In case of any further problems with the setup fill free to open new issues.

closed 2021-02-11 by jedrzejboczar