antmicro/rowhammer-tester #51: Reading CSR on ZCU104 is returning 0xFF

rolin-ron, 2021-02-17:
When running the dump_regs.py (after entering python virtual env.) I'm seeing this:
```
(venv) $ python dump_regs.py 
Using generated target files in: ../../build/zcu104
0x82000000: 0xffffffff ctrl_reset
0x82000004: 0xffffffff ctrl_scratch
0x82000008: 0xffffffff ctrl_bus_errors
0x82002000: 0xffffffff uart_rxtx
0x82002004: 0xffffffff uart_txfull
0x82002008: 0xffffffff uart_rxempty
0x8200200c: 0xffffffff uart_ev_status
...
0x82003800: 0xffffffff ddrphy_rst
0x82003804: 0xffffffff ddrphy_en_vtc
0x82003808: 0xffffffff ddrphy_half_sys8x_taps
0x8200380c: 0xffffffff ddrphy_wlevel_en
0x82003810: 0xffffffff ddrphy_wlevel_strobe
0x82003814: 0xffffffff ddrphy_cdly_rst
0x82003818: 0xffffffff ddrphy_cdly_inc
...
0x82005000: 0xffffffff ddrctrl_init_done
0x82005004: 0xffffffff ddrctrl_init_error
...
```
Which is a problem because when I go to run:
```
(venv) $ python mem.py 
Using generated target files in: ../../build/zcu104
 === Waiting for CPU to initialize DRAM ===
```
It gets stuck in a loop because it receives `0xFFFFFFFF` for `ddrctrl_init_done`. I know the communication with the EtherBone on the ZCU104 is working:
```
(venv) $ python etherbone_perf.py memwrite 32768 --burst 128
Using generated target files in: ../../build/zcu104
Elapsed = 0.097 sec
Size    = 128.000 KB
Speed   = 1.290 MBps
```
I can't seem to dump values of the CSR, and ideas on what is going on?

jedrzejboczar, 2021-02-17:
Hi, this is most likely due to a mismatch between CSR addresses in the bitstream on the board and the values used by the scripts. I should have added that when using the  prebuilt image you need to build a fresh bitstream and update it on the SD card. You need to build the bitstream (with `export TARGET=zcu104` and optional `export IP_ADDRESS=...` if you use a different one than the default), then copy it to the BOOT partition on the SD card and try once again. If it works you should be able to see the following value of `ctrl_scratch`:
```
0x82000004: 0x12345678 ctrl_scratch
```

We are working on improving the documentation to make the ZCU104 setup easier. [Here](https://github.com/antmicro/litex-rowhammer-tester/blob/migrate-to-rtd/doc/zcu104.rst#loading-bitstream) you can find an initial version of the instructions. We would be grateful for feedback if you find anything that should be there but is missing :)

jedrzejboczar, 2021-05-13:
Improved ZCU104 setup documentation can found at https://litex-rowhammer-tester.readthedocs.io/en/latest/zcu104.html along with general instructions https://litex-rowhammer-tester.readthedocs.io/en/latest/general.html. Closing this issue, if the problem reappears we can reopen.

closed 2021-05-13 by jedrzejboczar