antmicro/rowhammer-tester #74: Unable to probe Etherbone server at 192.168.100.50

LeonardooAlves, 2021-11-29:
Hi, I hope you are well.

I am trying to test the Rowhammer on the Arty A7 100T board, but when I try `export IP_ADDRESS=192.168.100.50 && make srv`, it returns the following error: `Exception: Unable to probe Etherbone server at 192.168.100.50.`

I have built the bitstream using the following commands: 
'export TARGET=arty
export IP_ADDRESS=192.168.100.50
make build TARGET_ARGS="--variant a7-100"`

I manually saved the bitstream to the board's flash memory using vivado, because the `make flash TARGET_ARGS="--variant a7-100"` command did not work.

After that, with both the USB and Ethernet cables connected to the computer, I tried: 
`export TARGET=arty
export IP_ADDRESS=192.168.100.50
make srv`

and the error occurs.

Also, when trying to simulate, I can't create the TUN interface or configure the IP address of the interface (not even with sudo).

Any ideas of what I am doing wrong?

Thank you for your help.

acomodi, 2021-11-30:
Hi @LeonardooAlves, you may need to configure the network accordingly before running `make srv`.

You can run the following:

```
sudo ip addr add 192.168.100.1/24 dev <eth card>
```

where `<eth card>` is the ethernet device you are using (you can check the devices with `ifconfig`). As an example, in my system, if I connect directly to the ethernet port of my host PC, the device name is `eno1`.

LeonardooAlves, 2021-12-07:
> Hi @LeonardooAlves, you may need to configure the network accordingly before running `make srv`.
>
> You can run the following:
>
> ```
> sudo ip addr add 192.168.100.1/24 dev <eth card>
> ```
>
> where `<eth [...]

Sorry for the late reply and the dumb questions. 

I can't run any command that requires `sudo` on the virtual environment:
```
(venv) leo@office:~/Documents/rowhammer/litex-rowhammer-tester$ tunctl -u $USER -t litex-sim
TUNSETIFF: Device or resource busy
(venv) leo@office:~/Documents/rowhammer/litex-rowhammer-tester$ sudo ip addr add 192.168.100.1/24 dev enp6s0
RTNETLINK answers: Operation not permitted
```

```
(venv) leo@office:~/Documents/rowhammer/litex-rowhammer-tester$ sudo ifconfig litex-sim 192.168.100.1/24 up
SIOCSIFADDR: Operation not permitted
SIOCSIFFLAGS: Operation not permitted
SIOCSIFFLAGS: Operation not permitted
SIOCSIFNETMASK: Operation not permitted
```
Therefore, I used the above commands outside the virtual environment (and it worked, that is, I did not have the operation not permitted message anymore). However, I still have the same error: `Exception: Unable to probe Etherbone server at 192.168.100.1/24.`

SeyoonChung, 2022-01-10:
> Sorry for the late reply and the dumb questions.
>
> I can't run any command that requires `sudo` on the virtual environment:
>
> ```
> (venv) leo@office:~/Documents/rowhammer/litex-rowhammer-tester$ tunctl [...]

Hi, @LeonardooAlves and @acomodi , I have a same problem above.
Is there any updates? Have you figured out what's wrong?

Here is my error log:
```c++
litex_server --udp --udp-ip 192.168.100.50 --udp-port 1234
[CommUDP] ip: 192.168.100.50 / port: 1234 / tcp port: 1234
Traceback (most recent call last):
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 52, in probe
    assert packet.pr == 1
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/venv/bin/litex_server", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'litex_server')()
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/third_party/litex/litex/tools/litex_server.py", line 270, in main
    server.open()
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/third_party/litex/litex/tools/litex_server.py", line 90, in open
    self.comm.open()
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 31, in open
    self.probe(self.server, self.port)
  File "/home/rhtest/toolkit/litex-row-hammer/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 57, in probe
    raise Exception(f"Unable to probe Etherbone server at {self.server}.")
Exception: Unable to probe Etherbone server at 192.168.100.50.
Makefile:63: recipe for target 'srv' failed
make: *** [srv] Error 1
```

My target board is "zcu104", by the way.
I'm looking forward to hearing from you.
Thanks.

notepractice, 2022-01-10:
The situation as above also happened in my environment. I think some additional options would be applied.
Is there any solution?

LeonardooAlves, 2022-01-20:
I can add the IP (outside the virtual environment only):
`sudo ip addr add 192.168.100.50 dev <eth card>`
Then, I can also ping it. However, I still cannot successfully run the `make srv` command, as the same error happens. I could not resolve it yet.

None of the commands that requires `sudo` works on the virtual environment; the docs mention something about it but I did not quite understand: "The repository contains a wrapper script around `sudo` which disallows LiteX to interfere with the host network configuration".

hbelatikar, 2022-03-21:
Is there any development on a solution for this issue?

acomodi, 2022-05-17:
Hi, I am quite sure that this is a network setup error.

What has been tested and proven to be working is a setup involving the following hardware:
- arty board (or the corresponding board under test e.g. zcu104, lpddr4_test_board)
- USB-ethernet adapter connected to the host USB port and the ethernet side connected to the board's ethernet port

At this point the ethernet adapter can be found on the host network configuration and by running `ifconfig` you can verify its presence there and get its name to be used as the `<eth interface>` in the next step.

With the bitstream already loaded on the device, the following commands can be used to create the litex server:
```
sudo ip addr add 192.168.100.1/24 dev <eth interface>
source venv/bin/activate
make srv
```

These steps have been performed on a clean setup and verified to be working both on arty and zcu104.

TommyCox, 2022-06-07:
> I resolved this issue by recompiling the zcu104_etherbone binary.

I also had this problem with the latest image for the zcu104, and it was also resolved by recompiling the etherbone binary.
I installed gcc-aarch64-linux-gnu via `apt` and was able to recompile it with `make LDFLAGS=-static` before using `scp` to push it to a running instance.

closed 2022-09-20 by tmichalak