antmicro/rowhammer-tester #77: Edit button on ReadTheDocs doesn't seem to work

mithro, 2022-01-05:
The Edit button found on most of the ReadTheDocs pages for this repository doesn't seem to work, it just takes you back to the same page;
[image]

kgugala, 2022-01-05:
I've removed the Edit button and replaced it with:
[image]

closed 2022-01-05 by kgugala

mithro, 2022-01-05:
The Edit link was suppose to take you directly to the file in GitHub repository...

pgielda, 2022-01-05:
Tbh the edit button was a leftover from our internal system...

mithro, 2022-01-05:
"Edit" links are quite common on ReadTheDocs sites - https://docs.pylonsproject.org/projects/pyramid/en/master/ 

https://docs.readthedocs.io/en/stable/guides/remove-edit-buttons.html
https://stackoverflow.com/a/66238784/3824056