antmicro/rowhammer-tester #100: Is it possible to generate the files for the ZCU102 board?

LeonardooAlves, 2022-05-03:
Hi, I hope you are well.

I noticed there are files (such as constraints and others) for the ZCU102 board. I wonder if it is possible to modify the ZCU104 scripts to replicate the tests on the ZCU102. Would the LiteDRAM work on it?

LeonardooAlves, 2022-05-03:
> What's your plan? Are you thinking of using the ZCU102's baked-in memory controller and PS-side SODIMM slot with the Rowhammer scripts that are in the repo?

I would like to exactly replicate the [zcu104](https://litex-rowhammer-tester.readthedocs.io/en/latest/zcu104.html#zcu104-chapter) implementation and tests (as the boards are very similar). Therefore, I think it uses the PS-side SODIMM with a PC > PS > PL connection.

jaccharrison, 2022-05-03:
I do not believe it is possible to replicate the ZCU104 implementation. The SODIMM slot on the ZCU102 is connected to the Zynq's processing system, not the programmable logic. Unless I am mistaken, this means that you can't create a LiteDRAM memory controller in the PL fabric and connect it to the SODIMM. 

It may be possible to replace LiteDRAM and its DFI interface in the RowHammer framework with a connection to the Zynq's baked-in memory controller. I haven't looked closely at this, however, so I can't give an educated opinion on how complex these changes would be.

LeonardooAlves, 2022-05-03:
> I do not believe it is possible to replicate the ZCU104 implementation. The SODIMM slot on the ZCU102 is connected to the Zynq's processing system, not the programmable logic. Unless I am mistaken, [...]

I may have understood it wrong, but the ZCU104 example described in the Docs uses the Zynq MPSoC to set up the Etherbone communication only. The LiteDRAM, CSR registers, etc., are all implemented on PL (FPGA) and connected to the FPGA DDR4. Hence, I thought I could replicate on the ZCU102 by just updating a few settings, such as the board constraints and FSBL.

LeonardooAlves, 2022-05-03:
> What DRAM are you testing on the ZCU102? Are you hoping to test DDR4 SODIMMs?

The 512MB 16-bit DDR4 attached to programmable logic (PL).

jaccharrison, 2022-05-03:
Ah, I assumed you were trying to test SODIMMs. In that case, what you are proposing might work.

LeonardooAlves, 2022-05-03:
> Ah, I assumed you were trying to test SODIMMs. In that case, what you are proposing might work.

Sorry for not being clear. Do you think the board constraints and FSBL are all I need to update to make it work? Or am I missing something here?

acomodi, 2022-05-04:
Hi @LeonardooAlves and @jaccharrison. Looking at the `zcu102` specs, I believe that this should be doable.

What would be needed is to:
- Enhance the [`zcu102` board platform](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_zcu102.py.) definition in litex-boards. Similarly to the [zcu104](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_zcu104.py), adding the `ddr4` pinout specification that needs to be adapted for the `zcu102` pinout.
- Verifying that the DDR4 module are the same as for the `zcu104`, otherwise this would need to be changed and, if the module definition is not present, it should be added alongside with the others in the [modules.py](https://github.com/enjoy-digital/litedram/blob/master/litedram/modules.py) of litedram.
- Add the `zcu102` target in this repo to generate the SoC. I believe this will be much similar to the `zcu104` one.

LeonardooAlves, 2022-05-04:
> Hi @LeonardooAlves and @jaccharrison. Looking at the `zcu102` specs, I believe that this should be doable.
>
> What would be needed is to:
>
> * Enhance the [`zcu102` board [...]

Thank you very much for your help.
I will add the DDR4 specification on the [board platform](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_zcu102.py) following the ZCU104 example.
Looking at the [modules.py](https://github.com/enjoy-digital/litedram/blob/master/litedram/modules.py), I can see that both ZCU102 DDR4 are available (the MT40A256M16 DDR4 chip and the MTA4ATF51264HZ SO-DIMM, which are connected to the PL and PS, respectively).

I will let you know how it goes. If it works, maybe, we can add it to the repository.

QZ-413, 2023-09-08:
> Thank you very much for your help. I will add the DDR4 specification on the [board platform](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_zcu102.py) following [...]

Hello, I encountered the same problem as you when trying to use zcu102. May I ask if the method mentioned above is feasible and has been successful?