antmicro/rowhammer-tester #101: LPDDR4 Ethernet/Etherbone not working

aaronachang, 2022-05-03:
Hello, I'm using the lpddr4 test board to try and run the Row Hammer test, but the only commands that I have gotten to run successfully are "make build" and "make upload". After running both commands, I am unable to connect to Etherbone, and I also noticed that the LEDs on the Ethernet port don't light up. I have gotten the Row Hammer test to work on the Arty A7 board, so I believe I should have the flow set up correctly.

I did notice that during synthesis during "make build", there were two critical warnings related to the Ethernet clocks, is this expected/normal?:

`---------------------------------------------------------------------------------
Start Applying XDC Timing Constraints
---------------------------------------------------------------------------------
CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 269 of /home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc. [/home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc:269]
CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 271 of /home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc. [/home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc:271]
WARNING: [Synth 8-3321] set_false_path : Empty through list for constraint at line 284 of /home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc. [/home/ssrl01/cindy/litex-rowhammer-tester/build/lpddr4_test_board/gateware/antmicro_lpddr4_test_board.xdc:284]`

acomodi, 2022-05-04:
Hi @aaronachang, to double check, are you using the same board and testbed described [here](https://litex-rowhammer-tester.readthedocs.io/en/latest/lpddr4_tb.html)?

If not, what hardware are you currently using for the LPDDR4 bitstream?

Those critical warnings are indeed hinting to the fact that the ethernet module might not be correctly configured, hence not working upon loading the bitstream.
I have just tried to run `make build` on my local machine and I couldn't reproduce the same critical warnings though, so it would be helpful if you could provide the exact steps that you performed, including the `litex-rowhammer-tester` version.

aaronachang, 2022-05-04:
Yes, I am using the same board and testbed as in the link you posted.

Here are the steps I followed (after verifying that I've pulled the latest version from git and verifying jumper configuration):
`source venv/bin/activate`
`source /home/ssrl01/labkits/Xilinx/2022.1/Vivado/2022.1/settings64.sh`
`export TARGET=lpddr4_test_board`
`make build`
`make upload`

aaronachang, 2022-05-07:
It seems like the critical warnings I got was a Vivado issue. Using Vivado 2017.3 instead of 2022.1 seemed to have removed the critical warnings.

However, I'm still not able to connect to the LPDDR4 board. I'm getting a similar error as #74
```
litex_server --udp --udp-ip 192.168.100.50 --udp-port 1234
[CommUDP] ip: 192.168.100.50 / port: 1234 / tcp port: 1234
Traceback (most recent call last):
  File "/home/ssrl01/cindy/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 52, in probe
    assert packet.pr == 1
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ssrl01/cindy/litex-rowhammer-tester/venv/bin/litex_server", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'litex_server')()
  File "/home/ssrl01/cindy/litex-rowhammer-tester/third_party/litex/litex/tools/litex_server.py", line 270, in main
    server.open()
  File "/home/ssrl01/cindy/litex-rowhammer-tester/third_party/litex/litex/tools/litex_server.py", line 90, in open
    self.comm.open()
  File "/home/ssrl01/cindy/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 31, in open
    self.probe(self.server, self.port)
  File "/home/ssrl01/cindy/litex-rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 57, in probe
    raise Exception(f"Unable to probe Etherbone server at {self.server}.")
Exception: Unable to probe Etherbone server at 192.168.100.50.
Makefile:63: recipe for target 'srv' failed
make: *** [srv] Error 1
```

aaronachang, 2022-05-19:
Turned out to be a hardware issue preventing the LPDDR4 board from being programmed correctly.

closed 2022-05-19 by aaronachang