antmicro/rowhammer-tester #103: Memtest fails during `mem.py`.

KimHwon, 2022-05-13:
Dear developers,
We're totally new in this hardware world, sorry if these questions might look obvious to you.

We've been working with ZCU104. While running `mem.py`, LiteX fails in memtest step. We got this error:

* Termial 1.
```
(venv) $ make sim
python rowhammer_tester/targets/zcu104.py --build --sim --ip-address 192.168.100.50 --mac-address 0x10e2d5000001 --udp-port 1234 
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-05-13 20:13:23)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 125.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv added.
INFO:SoC:CPU vexriscv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU vexriscv overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoC:CPU vexriscv adding Interrupt(s).
INFO:SoC:CPU vexriscv adding SoC components.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00010000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00010000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoCCSRHandler:sim_trace CSR allocated at Location 0.
INFO:SoCCSRHandler:sim_marker CSR allocated at Location 1.
INFO:SoCCSRHandler:sim_finish CSR allocated at Location 2.
INFO:SoCCSRHandler:leds CSR allocated at Location 3.
INFO:SoC:Using DRAM module MTA4ATF51264HZ ratio 1:4
Traceback (most recent call last):
  File "rowhammer_tester/targets/zcu104.py", line 279, in <module>
    main()
  File "rowhammer_tester/targets/zcu104.py", line 269, in main
    soc = SoC(**soc_kwargs)
  File "rowhammer_tester/targets/zcu104.py", line 188, in __init__
    super().__init__(**kwargs)
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/rowhammer_tester/targets/common.py", line 163, in __init__
    verbosity = 3,
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/third_party/litedram/litedram/phy/model.py", line 576, in __init__
    verbose      = verbosity > SDRAM_VERBOSE_DBG)
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/third_party/litedram/litedram/phy/model.py", line 236, in __init__
    self.prepare_timings(timings, refresh_mode, memtype)
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/third_party/litedram/litedram/phy/model.py", line 211, in prepare_timings
    val = val[refresh_mode]
TypeError: tuple indices must be integers or slices, not str
Makefile:37: recipe for target 'sim' failed
make: *** [sim] Error 1
```

* Termial 2.
```
(venv) $ python mem.py
Using generated target files in: ../../build/zcu104
 === Waiting for CPU to initialize DRAM ===
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on May 13 2022 19:21:50
 BIOS CRC passed (42d2439d)

 Migen git sha1: ac70301
 LiteX git sha1: 80b309d1

--=============== SoC ==================--
CPU:		VexRiscv_Min @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 49.5MiB/s
   Read speed: 52 === Initialization succeeded. ===
Proceeding ...

Memtest (basic)
Traceback (most recent call last):
  File "mem.py", line 89, in <module>
    run_memtest('basic', itertools.cycle([0xaaaaaaaa, 0x55555555]))
  File "mem.py", line 79, in run_memtest
    errors = memtest(wb, length=memtest_size, generator=generator, **kwargs)
  File "mem.py", line 23, in memtest
    data = memread(wb, length, base=base, burst=burst)
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/rowhammer_tester/scripts/utils.py", line 173, in memread
    data += wb.read(base + 4 * i, min(burst, n - i))
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/third_party/litex/litex/tools/litex_client.py", line 67, in read
    packet.decode()
  File "/home/scalar4/source/zcu104/litex-rowhammer-tester/third_party/litex/litex/tools/remote/etherbone.py", line 306, in decode
    header = list(ba[:etherbone_packet_header.length])
TypeError: 'int' object is not subscriptable
```

We tried with Samsung M471A5244CB0-CTD(DDR4) and Samsung M471A1G44AB0-CWE(DDR4), but still failed. And also, We maually built boot image. It failed again. Lastly, We checked network setup with Wireshark but cannot found any problems.

We accessed FPGA to check Etherbone, surprisingly, /bin/zcu104_etherbone was stuck even though they sending or receiving packets. This issue also occurred when DEBUG is enabled in firmware/zcu104/etherbone/Makefile.
```
# /bin/zcu104_etherbone 
Serving on port 1234 ...
```

Thank you for your explanations.

acomodi, 2022-05-17:
Hi @KimHwon thanks for filing this issue.

One thing I notice is that you are running the `make sim` target, which I think was not tested for the zcu104 target and I can confirm it is failing for me as well and I get the same result.

If the intent was to test on hardware directly though, I think that you can probably grab the latest release that contains the firmware to successfully configure the PS to make the bridge between the Ethernet port and the Programmable Logic.

The latest release can be found here: https://github.com/antmicro/litex-rowhammer-tester/releases/tag/zcu104-v0.3

Once you download the latest release, you can follow the guide here: https://litex-rowhammer-tester.readthedocs.io/en/latest/zcu104.html.

If you want to rebuild the bitstream, you can do so by running:

```
TARGET=zcu104 make build
```

Once the bitstream is generated, you can overwrite the bitstream present in the BOOT partition of the SD and it should work as well.

KimHwon, 2022-05-17:
Thank you for answer.

We fixed this issue.
But now, #74 happened to us.

acomodi, 2022-05-17:
I have added some more detailed instructions in the [latest comment](https://github.com/antmicro/litex-rowhammer-tester/issues/74#issuecomment-1128756805).

Can you please describe what is your current setup?

It would be helpful to understand:
- the `ifconfig` output to see the network setup
- the hardware setup

Furthermore, are the leds correctly blinking on the zcu104 board, as well as the ethernet-related leds?

KimHwon, 2022-05-18:
We configurated it as follows:

**1. With local network**
```
+------+     +--------+      +------+
|  PC  +=====+ Router +======+ FPGA |
+------+     +--------+      +------+
```

* PC
IP address : 192.168.100.1/24 (enp2s0)

* ZCU104
IP address : 192.168.100.50/24

* PC and ZCU104 connet to same Router.
Local network is 192.168.100.0/24.

```bash
enp2s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.1  netmask 255.255.255.0  broadcast 192.168.100.255
        inet6 fe80::9c1:5851:1502:629a  prefixlen 64  scopeid 0x20<link>
        ether 9c:7b:ef:3c:e5:89  txqueuelen 1000  (Ethernet)
        RX packets 2460751  bytes 2752641389 (2.7 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1508633  bytes 195506727 (195.5 MB)
        TX errors 9  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3151817  bytes 263282351 (263.2 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3151817  bytes 263282351 (263.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

**2. With direct connection**
```
+-----+       +------+
| P C +=======+ FPGA |
+-----+       +------+
```

* PC
IP address : 192.168.100.1/24 (enp2s0)
default GW : 192.168.100.50

* ZCU104
IP address : 192.168.100.50/24
default GW : 192.168.100.1/24

```bash
enp2s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.1  netmask 255.255.255.0  broadcast 192.168.100.255
        inet6 fe80::9c1:5851:1502:629a  prefixlen 64  scopeid 0x20<link>
        ether 9c:7b:ef:3c:e5:89  txqueuelen 1000  (Ethernet)
        RX packets 2461962  bytes 2753466304 (2.7 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1509618  bytes 195662014 (195.6 MB)
        TX errors 9  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3151985  bytes 263299605 (263.2 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3151985  bytes 263299605 (263.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Both case, PC and ZCU104 passed ping test to each other.
Of course, LEDs which related to loading bitstream successfully is blink in a circular pattern. And DS27 LED is green which related to Ethernet link.

We think some additional options would be applied, too.

p.s. If possible, it would be helpful if you could let us know the successful setting.

acomodi, 2022-05-18:
A successful setting in my case that I have tested to be working is the following:
```
PC --> USB-Ethernet Adapter --> ZCU104
```

A successful setup will result in the following output when pinging the gateware from the PC:
```
ping 192.168.100.50
PING 192.168.100.50 (192.168.100.50) 56(84) bytes of data.
64 bytes from 192.168.100.50: icmp_seq=1 ttl=64 time=1.08 ms
64 bytes from 192.168.100.50: icmp_seq=2 ttl=64 time=0.519 ms
64 bytes from 192.168.100.50: icmp_seq=3 ttl=64 time=0.530 ms
64 bytes from 192.168.100.50: icmp_seq=4 ttl=64 time=0.531 ms
64 bytes from 192.168.100.50: icmp_seq=5 ttl=64 time=0.542 ms
64 bytes from 192.168.100.50: icmp_seq=6 ttl=64 time=0.525 ms
64 bytes from 192.168.100.50: icmp_seq=7 ttl=64 time=0.500 ms
64 bytes from 192.168.100.50: icmp_seq=8 ttl=64 time=0.528 ms
64 bytes from 192.168.100.50: icmp_seq=9 ttl=64 time=0.518 ms
64 bytes from 192.168.100.50: icmp_seq=10 ttl=64 time=0.524 ms
64 bytes from 192.168.100.50: icmp_seq=11 ttl=64 time=0.519 ms
64 bytes from 192.168.100.50: icmp_seq=12 ttl=64 time=0.527 ms
64 bytes from 192.168.100.50: icmp_seq=13 ttl=64 time=0.515 ms
64 bytes from 192.168.100.50: icmp_seq=14 ttl=64 time=0.535 ms
...
```

If the above is achieved, than you can proceed with opening two terminals:

**Terminal 1**:

Run `make srv` from the repository root

**Terminal 2**:

Run `./mem.py` script to train the memory

To better understand your setup, what image are you using and how are you programming the device (both the PS and PL)?

I am currently testing using a microSD on which I have loaded the image from the latest release.

acomodi, 2022-05-18:
Great to hear that everything works fine! You may also update the bitstream if needed, probably the name of the file will need to be the same though (`zcu104.bit`).

Closing this issue for now as it is resolved.

closed 2022-05-18 by acomodi