antmicro/rowhammer-tester #144: spd_eeprom.py times out if memory training fails and cannot read the EEPROM

dwalton64, 2022-10-23:
[https://github.com/antmicro/rowhammer-tester/blob/master/rowhammer_tester/scripts/spd_eeprom.py ](https://github.com/antmicro/rowhammer-tester/blob/2d2b811a30878e54ed9c675ea736ed1fc1d5249c/rowhammer_tester/scripts/spd_eeprom.py#L109)  waits until memory training **successfully** completes before reading the SPD EEPROM.

This is a problem when the training fails as in the example below:
```  m15, b04: |00111111111111110000000000000000| delays: 09+-07
  m15, b05: |00000000000000000001111111111111| delays: 25+-06
  m15, b06: |00000000000000000000000000000000| delays: -
  m15, b07: |00000000000000000000000000000000| delays: -
  best: m15, b04 delays: 09+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  0/256
  addr errors: 0/8192
  data errors: 6899/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> 
litex> sdram_spd 0
Memory dump:
0x00000000  23 12 0c 01 85 29 00 08 00 60 00 03 08 0b 80 00  #....)...`......
0x00000010  00 00 06 0d f8 3f 00 00 6e 6e 6e 11 00 6e f0 0a  .....?..nnn..n..
0x00000020  20 08 00 05 00 60 18 28 2b 00 78 00 14 3c 00 00   ....`.(+.x..<..
0x00000030  00 00 00 00 00 00 00 00 00 00 00 00 16 16 15 16  ................
0x00000040  03 16 03 16 03 16 03 16 0d 16 16 16 16 16 00 00  ................
0x00000050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000070  00 00 00 00 00 ec 9c 00 00 00 00 00 e7 00 24 97  ..............$.
0x00000080  11 11 41 19 00 86 32 a0 01 65 05 00 00 00 00 00  ..A...2..e......
0x00000090  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000a0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000b0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000c0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000d0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000e0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x000000f0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b e4  ..............{.
```
The example above will cause spd_eeprom.py to timeout since the training fails.  But it also shows that it is still possible to read the SPD data.

The code needs to wait for the training to complete.  But it should read the SPD EEPROM regardless of whether the training is successful or not.

tmichalak, 2023-01-18:
@dwalton64 we have just merged a [PR](#164) that should help in your case. Just to be clear, reading the SPD does not depend on the result of mem_test, but rather how long it takes. The script waits for the BIOS prompt before it executes the `read` command and this is where the timeout happened in your example. Previously there was a customizable timeout limit in the `spd_eeprom.py` script which we removed and now simply wait until the training is done (regardless of the result).

tmichalak, 2024-01-11:
This has been fixed. Closing.

closed 2024-01-11 by tmichalak