antmicro/rowhammer-tester #152: Running mem_test at the BIOS console hangs if the memory range is too large.

dwalton64, 2022-12-22:
Running python bios_console.py, I ran a memory test after sucessful memory initialization and saw it hang when the memory size was greater that 4 Gbytes:

```
litex> mem_test 0x40000000 0x100000000
Memtest at 0x40000000 (3.10GiB)...
  Write: 0x40000000-0x80000000 1.0GiB
```

The memory test above hung and I had to reset the board.  The terminal stopped echoing the things I typed. 

Chatting with the Antmicro folks, it sounds like the internal CPU is a 32 bit CPU.  The DRAM on the DIMM is mapped at 0x40000000 and thus we can't test 4 Gbytes.  

It would be useful if the software checked the range being tested and prevented the hang.  It would be better if the memory was accessed through a window or a MMU so that we can test any address in memory.  This would be especially useful in the case where we see an unexpected error from a rowhammer attack and we want to validate that the memory at the address of the error is working correctly.

michalsieron, 2023-01-20:
Unfortunately, `mem_test` command in the LiteX BIOS is a general command for testing any memory and not only RAM, so we can't check its range against the `MAIN_RAM` region.

The hang doesn't come from the size being greater than 4 GiB, but rather from memory operations on addresses that don't belong to any defined memory region.
As the CPU performs multiple invalid writes/reads, it gets [timed out](https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L241) and each next operation is delayed by 0.1 s.

If you want to test memory using physical addresses, you can use the newly added command `sdram_hw_test` (#159).

Like `mem_test`, it requires two arguments: `origin` and `size`.
Optionally, you can pass `burst_length` which is a number of full writes to the DRAM before memory content is checked.

As it is counted in DRAM transfers, it depends on the bus width. Currently, one transfer is 32 bytes as we use only the lower 8 modules for DDR4 memories.

So to test 4 GiB like in your example, you would run:
```
litex> sdram_hw_test 0x0 0x100000000 0x2000000
```
This will test first 4 GiB of memory in 1 GiB chunks.

tmichalak, 2024-01-11:
The [PR](#159) which adds the `sdram_hw_test` command has been merged. Closing.

closed 2024-01-11 by tmichalak