antmicro/rowhammer-tester #185: # Inconsistent Retention Errors with Refresh Disabled Before Memory Writes

biecho, 2024-10-05:
## Description

I observed an unexpected behavior when testing retention errors by disabling DRAM refresh. When I disable refresh **after** writing to a row, the expected retention errors occur, but when I disable refresh **before** the write, no errors are observed, even after significantly increasing the delay.

### Code Examples and Behavior

#### Working Case (Errors Observed)
In the code snippet below, I disable refresh **after** writing data to a row and hold the data for 10 seconds. When reading back, I see expected retention errors:

```python
# Write some data to the column we are reading to check that scratchpad gets filled
converter = DRAMAddressConverter.load()
data = [0xffffffff] * 128
memwrite(wb, data, base=converter.encode_bus(bank=0, row=row, col=0))

wb.disable_refresh()
time.sleep(10)
wb.enable_refresh()
```

**Output:**
```
Scratchpad contents:
0x00000000:  ff ef ff ff 7f ff ff ef df bf ff ff ff ff ff bf  ................
0x00000010:  fb fe ff ff ff 7f ff ff ff ff ff ff ff f7 ff ff  ................
0x00000020:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000030:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

#### Failing Case (No Errors Observed)
However, if I disable refresh **before** the write, and then hold for the same period (10 seconds or even longer), I see no errors at all:

```python
# Write some data to the column we are reading to check that scratchpad gets filled
converter = DRAMAddressConverter.load()
data = [0xffffffff] * 128
wb.disable_refresh()
memwrite(wb, data, base=converter.encode_bus(bank=0, row=row, col=0))

time.sleep(10)
wb.enable_refresh()
```

**Output:**
```
Scratchpad contents:
0x00000000:  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0x00000010:  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
0x00000020:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x00000030:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

### Expected Behavior

Given that the DRAM refresh is disabled for an extended period, I would expect the data written in the row to degrade over time due to retention errors, especially with a hold time of up to a full minute. Instead, I see no errors, indicating that all the bits remain intact.

### Possible Issue

It appears that disabling the refresh before the data is written might be affecting the internal state of the memory subsystem, leading to unexpected behavior. This could suggest that disabling the refresh is influencing the logic of the platform beyond strictly stopping refresh operations, possibly impacting other signals or internal mechanisms that maintain the data's integrity.

### Steps to Reproduce

1. Use the initial script setup that writes a known pattern (`0xffffffff`) to a row.
2. Disable refresh before the write and wait for 10 seconds.
3. Read back the contents and verify if the pattern remains intact.
4. Compare this with the behavior when disabling refresh **after** the write.

Does disabling refresh influence internal signals or logic beyond simply stopping refresh operations?

The full script is here:

```
#!/usr/bin/env python3

import time

from rowhammer_tester.gateware.payload_executor import Encoder, OpCode
from rowhammer_tester.scripts.utils import memdump, memread, memwrite, DRAMAddressConverter, RemoteClient, read_ident, \
    get_litedram_settings

# Sample program

encoder = Encoder(bankbits=5, nranks=2)
row = 0
PAYLOAD = [
    encoder(OpCode.NOOP, timeslice=50),
    encoder(OpCode.PRE, timeslice=20, address=encoder.address(rank=0, bank=0)),
    encoder(OpCode.ACT, timeslice=20, address=encoder.address(rank=0, bank=0, row=row)),
    encoder(OpCode.READ, timeslice=20, address=encoder.address(rank=0, bank=0, col=0 | (1 << 11))),
    encoder(OpCode.NOOP, timeslice=50),
]

def execute(wb):
    program = [w for w in PAYLOAD]
    program += [0] * (wb.mems.payload.size // 4 - len(program))  # fill with NOOPs

    # Write some data to the column we are reading to check that scratchpad gets filled
    converter = DRAMAddressConverter.load()
    data = [0xffffffff] * 128
    memwrite(wb, data, base=converter.encode_bus(bank=0, row=row, col=0))

    wb.disable_refresh()
    time.sleep(10)
    wb.enable_refresh()

    print('\nTransferring the payload ...')
    memwrite(wb, program, base=wb.mems.payload.base)

    def ready():
        status = wb.regs.payload_executor_status.read()
        return (status & 1) != 0

    print('\nExecuting ...')
    assert ready()
    wb.regs.payload_executor_start.write(1)
    while not ready():
        time.sleep(0.001)

    print('Finished')

    print('\nScratchpad contents:')
    scratchpad = memread(wb, n=16, base=wb.mems.scratchpad.base)
    memdump(scratchpad, base=0)

if __name__ == "__main__":
    wb = RemoteClient()
    wb.open()
    print("Board info:", read_ident(wb))

    execute(wb)

    wb.close()

```

mtdudek, 2024-10-05:
Hi,
Could you run this test for me?
```
converter = DRAMAddressConverter.load()
data = [0xffffffff] * 128

self.regs.controller_settings_refresh.write(0)

memwrite(wb, data, base=converter.encode_bus(bank=0, row=row, col=0))
memwrite(wb, data, base=converter.encode_bus(bank=0, row=(row+1), col=0))

time.sleep(10)
self.regs.controller_settings_refresh.write(1)
```
I wonder if disabling refresh cases the row to stay opened,
with its contents still in the SRAM buffer.

biecho, 2024-10-05:
Hi @mtdudek, thanks for the very fast correct answer 👍  It does make a difference. With your code, it works again.

biecho, 2024-10-07:
A quick follow-up question: Is the SRAM buffer the L2 cache, whose size is configured at build time?

mtdudek, 2024-10-07:
SRAM buffer is a part of the DRAM data circuitry.
During ROW activation data is set on the column lines, which are then amplified using analog amplifiers and these values are stored in the access buffer(SRAM)

closed 2024-10-09 by mtdudek