antmicro/rowhammer-tester #192: RDIMM flag on DDR5 devices

biecho, 2024-11-21:
When building the bitstream for an RDIMM DDR5 device, I noticed that the generated litedram_settings.json file includes the following entry:

```
"is_rdimm": false
```

This seems to impact the training of the device. Is this behavior a bug? Should we manually set this value when using the build scripts?

mtdudek, 2024-11-21:
Thanks for opening this issue.
I've checked Rowhammer code with its dependencies.
It turns out that this `is_rdimm` field is never used in the code base.
I'll remove this filed from the `litedram_settings.json`.

As for training code in the `third_party/litex/litex/soc/software/liblitedram/ddr5_training.c`.
It does have an `is_rdimm` variable, but it's being set by `read_module_type` function, which reads DIMMs SPD memory to check for attached memory type.

mtdudek, 2024-11-28:
I've modified code, so that `is_rdimm` field will not be added to `litedram_settings.json`.
I think this closes issue, but feel free to reopen it.

closed 2024-11-28 by mtdudek