antmicro/rowhammer-tester #196: Reading out SPD data fails

pjattke, 2024-12-06:
We have issues with reading out the SPD data of DDR5 RDIMMs with the DDR5 Tester platform using the `spd_eeprom.py` script. The DIMM we are using is a [M321R4GA3BB6-CQK](https://memory.net/product/m321r4ga3bb6-cqk-samsung-1x-32gb-ddr5-4800-rdimm-pc5-38400r-dual-rank-x8-module/). 

First, we had the issue that the CPU memory training never ended:
```
(venv) pjattke@ee-tik-cn126:~/git/rowhammer-tester/rowhammer_tester/scripts$ ./spd_eeprom.py read /data/pjattke/fpga/ddr5-rdimm-info/dimm_315.spd
Using generated target files in: ../../build/ddr5_tester
Reading SPD EEPROM ...
Board info:
[-] Waiting for CPU to finish memory training
```

This raises the first question: why is it trying to do a training if we don't have a valid bitstream for that device yet and we cannot generate a bitstream because it is not letting us to read out the SPD data?

We then figured out that we could use the bitstream of another similar DIMM (where we have the SPD data). This allowed us to successfully pass the DRAM training. Invoking `spd_eeprom.py` again, gives us another error:

```
(venv) pjattke@ee-tik-cn126:~/git/rowhammer-tester/rowhammer_tester/scripts$ export TARGET=ddr5_tester; ./spd_eeprom.py read dimm_315.spd
Using generated target files in: ../../build/ddr5_tester
Reading SPD EEPROM ...
Board info:
Ready
Traceback (most recent call last):
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/./spd_eeprom.py", line 146, in <module>
    output = read_spd(console, *SPD_COMMANDS[target])
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/./spd_eeprom.py", line 31, in read_spd
    console.expect(prompt)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/spawnbase.py", line 354, in expect
    return self.expect_list(compiled_pattern_list,
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/spawnbase.py", line 383, in expect_list
    return exp.expect_loop(timeout)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/expect.py", line 181, in expect_loop
    return self.timeout(e)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/expect.py", line 144, in timeout
    raise exc
pexpect.exceptions.TIMEOUT: <pexpect.pty_spawn.spawn object at 0x7f3f5adcadf0>
command: /local/home/pjattke/git/rowhammer-tester/venv/bin/python
args: ['/local/home/pjattke/git/rowhammer-tester/venv/bin/python', 'bios_console.py', '-t', 'litex_term']
buffer (last 100 chars): b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
before (last 100 chars): b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 2289696
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'^.*litex[^>]*> ')
<pexpect.pty_spawn.spawn object at 0x7f3f5adcadf0>
command: /local/home/pjattke/git/rowhammer-tester/venv/bin/python
args: ['/local/home/pjattke/git/rowhammer-tester/venv/bin/python', 'bios_console.py', '-t', 'litex_term']
buffer (last 100 chars): b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
before (last 100 chars): b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 2289696
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'^.*litex[^>]*> ')
```

Why is this script not working anymore? We have been using it in the past to successfully read out the SPD data of our RDIMMs.

mtdudek, 2024-12-06:
Hi, just a quick question.
Can you provide log of the training pass?

In the provided logs I see that board info is empty.
It also looks like you are receiving 0's  instead of valid data from the board.

As for why training is done before SPD is read.
It is because scripts use bios shell to issues I2C commands to access SPD.
This shell only becomes available after bios initialization is done,
and memory training is part of the initialization.

We are working on an utility that will allow for I2C accesses before the shell is active.

pjattke, 2024-12-06:
> Can you provide log of the training pass?
Yes sure, please find the training log in [training.txt](https://github.com/user-attachments/files/18038720/training.txt).

> In the provided logs I see that board info is empty. It also looks like you are receiving 0's instead of valid data from the board.
How is it possible that the memory training still passes although it seems like it is not receiving valid data from the board?

> We are working on an utility that will allow for I2C accesses before the shell is active.
Great, thanks! Is there an ETA for this new feature?

mtdudek, 2024-12-06:
In the provided training log we've noticed that bitstream was build from commit `6a22a6f7104e31c203d6fb823d91ba914bdba137`.
Since we can't find it in the RH repo, could you please rebuild your bitstream using current main branch and rerun `spd_eeprom.py`?

pjattke, 2024-12-06:
Yes, you are right. This private commit of our fork is based on your rowhammer-tester repo. I rebuilt it now for your commit on main branch (6a22a6f). 

Here's the log of the training: [training.txt](https://github.com/user-attachments/files/18040555/training.txt)

Running `spd_eeprom.py` then gives me following:

```
(venv) pjattke@ee-tik-cn126:~/git/rowhammer-tester/rowhammer_tester/scripts$ export TARGET=ddr5_tester; ./spd_eeprom.py read /data/pjattke/fpga/ddr5-rdimm-info/spd/dimm_315.spd
Using generated target files in: ../../build/ddr5_tester
Reading SPD EEPROM ...
Board info: Row Hammer Tester SoC on xc7k160tffg676-3, git: bd05e520f30fea3b554e495bb3a92d60d0a08c97 2024-12-06 15:57:26
Ready
Reading page 0
Traceback (most recent call last):
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/./spd_eeprom.py", line 148, in <module>
    output = read_spd(console, *SPD_COMMANDS[target])
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/./spd_eeprom.py", line 51, in read_spd
    console.expect(prompt)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/spawnbase.py", line 354, in expect
    return self.expect_list(compiled_pattern_list,
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/spawnbase.py", line 383, in expect_list
    return exp.expect_loop(timeout)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/expect.py", line 181, in expect_loop
    return self.timeout(e)
  File "/local/home/pjattke/git/rowhammer-tester/venv/lib/python3.9/site-packages/pexpect/expect.py", line 144, in timeout
    raise exc
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7fdbcc8f09d0>
command: /local/home/pjattke/git/rowhammer-tester/venv/bin/python
args: ['/local/home/pjattke/git/rowhammer-tester/venv/bin/python', 'bios_console.py', '-t', 'litex_term']
buffer (last 100 chars): b'ine 363, in send_packet\r\n    socket.sendall(packet.bytes)\r\nBrokenPipeError: [Errno 32] Broken pipe\r\n'
before (last 100 chars): b'ine 363, in send_packet\r\n    socket.sendall(packet.bytes)\r\nBrokenPipeError: [Errno 32] Broken pipe\r\n'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 2563133
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'^.*litex[^>]*> ')
```

In the other window, where I am running `make srv`, I can see the following error:

```
(venv) pjattke@ee-tik-cn126:~/git/rowhammer-tester$ make srv
litex_server --udp --udp-ip 192.168.100.50 --udp-port 1234
[CommUDP] ip: 192.168.100.50 / port: 1234 / tcp port: 1234
Connected with 127.0.0.1:48558
Connected with 127.0.0.1:48570
Disconnect
Disconnect
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/local/home/pjattke/git/rowhammer-tester/third_party/litex/litex/tools/litex_server.py", line 153, in _serve_thread
    reads += self.comm.read(addr, length, burst)
  File "/local/home/pjattke/git/rowhammer-tester/third_party/litex/litex/tools/remote/comm_udp.py", line 116, in read
    datas = record.writes.get_datas()
AttributeError: 'NoneType' object has no attribute 'get_datas'
```

mtdudek, 2024-12-06:
Is it possible that `third_party/litex` is out-of-date?

pjattke, 2024-12-09:
Thanks @mtdudek, this was indeed the issue. Updating the third-party dependencies to the latest version fixed the issue and we could successfully read out the SPD data. We are now building the bitstream to see if the device works.

pjattke, 2024-12-27:
Hello @mtdudek, reading out the SPD seems to be broken in the latest commit (dfabb627bc165f827174377dcba0991881904970). 

I am using the FPGA with S/N 019. I first run the `mem.py`:

```
(venv) pjattke@ee-tik-cn119:~/git/rowhammer-tester$ rowhammer_tester/scripts/mem.py --srv
Using generated target files in: build/ddr5_tester
tcp port: 1234
Connected with 127.0.0.1:51622
Board info: Row Hammer Tester SoC on xc7k160tffg676-3, git: bd05e520f30fea3b554e495bb3a92d60d0a08c97 2024-12-17 04:24:36
 === Waiting for CPU to initialize DRAM ===

... lots of output ...

Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x402 === Initialization succeeded. ===
Proceeding ...

Memtest (basic)
OK

Memtest (random)
OK
```

Then, trying to run `./spd_eeprom.py` gives me this error:

```
(venv) pjattke@ee-tik-cn119:~/git/rowhammer-tester$ rowhammer_tester/scripts/spd_eeprom.py read --srv out.spd
Using generated target files in: build/ddr5_tester
tcp port: 1234
Reading SPD EEPROM ...
Connected with 127.0.0.1:49748
Board info: Row Hammer Tester SoC on xc7k160tffg676-3, git: bd05e520f30fea3b554e495bb3a92d60d0a08c97 2024-12-17 04:24:36
Reading page 0
Traceback (most recent call last):
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/spd_eeprom.py", line 165, in <module>
    spd_data = read_spd_i2c(wb, *SPD_COMMANDS[target])
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/spd_eeprom.py", line 38, in read_spd_i2c
    i2c_write(wb, 0x50, 0x0B, [page], 1)
  File "/local/home/pjattke/git/rowhammer-tester/rowhammer_tester/scripts/utils.py", line 682, in i2c_write
    raise ValueError(error_msg)
ValueError: Requested write length(1) exceeds the number of available entries in the I2C Worker FIFO(-2
```

However, using commit bd05e520f30fea3b554e495bb3a92d60d0a08c97 seems to work fine:

```
(venv) pjattke@ee-tik-cn119:~/git/rowhammer-tester$ rowhammer_tester/scripts/spd_eeprom.py read --srv out.spd
Using generated target files in: build/ddr5_tester
tcp port: 1234
Reading SPD EEPROM ...
Connected with 127.0.0.1:43208
Board info: Row Hammer Tester SoC on xc7k160tffg676-3, git: bd05e520f30fea3b554e495bb3a92d60d0a08c97 2024-12-17 04:24:36
Ready
Connected with 127.0.0.1:43212
Disconnect
Reading page 0
Reading page 1
Reading page 2
Reading page 3
Reading page 4
Reading page 5
Reading page 6
Reading page 7
SPD saved to file: out.spd
```

mtdudek, 2024-12-27:
Can you rebuild your bitstream using newest mainline commit (dfabb627bc165f827174377dcba0991881904970)?
There was a change in the SPD/I2C scripts. DDR5 platform started to utilize HW I2C worker to speed up SPD read and allow for access before system boot sequence is done.

In the error message you provided, bitstream was build on commit (bd05e520f30fea3b554e495bb3a92d60d0a08c97), at that point i2C worker was not yet added.

pjattke, 2024-12-28:
This solved the issue. Thank you very much @mtdudek!

closed 2024-12-28 by pjattke