antmicro/rowhammer-tester #197: Bug Report: Instruction Timeslice Value Exceeds Max Allowed

biecho, 2024-12-14:
### **Description**
The following code snippet:

```python
encoder.Instruction(OpCode.REF, timeslice=timing.tRFC)
```

results in an assertion error:

```
AssertionError: Timeslice value: 60 exceeded max value: 31
```

### **Expected Behavior**
The `Instruction` with `timeslice=timing.tRFC` should work as expected, provided that `timing.tRFC` is valid. There seems to be a mismatch between the `timeslice` value and the maximum allowed limit.

### **Observed Behavior**
An assertion is raised, indicating that the `timeslice` value exceeds the maximum allowed (31):

```
AssertionError: Timeslice value: 60 exceeded max value: 31
```

### **Steps to Reproduce**
1. Use the `Instruction` method with the following parameters:
   ```python
   encoder.Instruction(OpCode.REF, timeslice=timing.tRFC)
   ```
2. Set `timing.tRFC` to a value greater than 31 (e.g., 60).
3. Observe the assertion error.

wkkuna, 2024-12-18:
Hi @biecho,

Thanks for the report.
We've taken a look at the issue and it seems to be an oversight - as up until the DDR4 the refresh would be executed under 32 system clock cycles.
Due to the fact that the DDR5 targets use a system clock of a higher frequency than the previous models, a larger number of cycles required for the refresh execution.

Due to the limitations of the instruction encoding and the fact that we're currently working on its improvements, the issue's been resolved by having the instrucion followed by an appropriate number of NOOPs to fulfill the requested `timeslice` parameter.

Once an instruction is issued with the `timeslice` that exceeds the number of bits dedicated in the internal representation (5) it is expected to be notified by a warning in the console log:
```
WARNING: Timeslice value:{requested timeslice} exceeded max value intended for instruction: 31
```

The remaining number of cycles will be filled out by the NOOP instructions.
This behavior provides the expected delay before the next instruction.

Please try with the newly at the released changes and let us know if this solves the issue in your use case.

tmichalak, 2024-12-27:
@biecho have you been able to rerun the tests with the latest changes?

biecho, 2025-01-06:
Thanks for fixing this. Seems to work again.

closed 2025-01-06 by tmichalak