antmicro/rowhammer-tester #198: Record cycle counter when payload began executing

sqazi, 2024-12-29:
It would be very useful to have a feature that records a cycle count value when a payload begins executing.  An internal cycle count register should increment continuously and be snapshotted into an accessible register whenever a payload begins executing.  The script can then use that value to make comparisons between multiple payload executions.

wkkuna, 2024-12-30:
Hi,
Sure thing, I'll have a look at it.

wkkuna, 2025-01-02:
Hi,

We've just released changes that include the Payload Executor CSR reporting the first and last cycle of the payload execution.
The register `payload_executor_exec_start` reports on the first system clock cycle (counting from the last power cycle) of the payload execution.
There's also `payload_executor_exec_stop` that snapshots the first cycle the payload has stopped being executed.

After rebuilding your design and running payload execution (e.g. via `hw_rowhammer.py`) you should see additional:

```
Payload execution started at: 33791782680
Payload execution finished at: 33922040349
Total execution time (system clock cycles): 130257669
```

in your logs.
Let us know if this works for you.