antmicro/rowhammer-tester #200: Prebuilt SD card image for ZCU104

kjh3141592, 2025-01-10:
Thank you for the great work.

I am trying to set SD card for ZCU104 board by following the document:
https://github.com/antmicro/rowhammer-tester/blob/main/docs/source/zcu104.md

 I cannot download the image file from the link below. It returns 404 not found error.
https://github.com/antmicro/rowhammer-tester/releases/tag/zcu104-v0.2

Please update the download link or let me know alternative way to download.

Thank you again for the great work.

mtdudek, 2025-01-10:
Thanks for reaching out.

We will look into creating new release with SD card image zip and update documentation to reflect this change.
In the meantime, documentation contains instruction on how to build SD image https://antmicro.github.io/rowhammer-tester/zcu104.html#zcu104-sd-card-image.

summerhot12138, 2025-11-19:
Thank you for the great work.

I followed the documentation to create the SD card image file:
https://antmicro.github.io/rowhammer-tester/zcu104.html#zcu104-sd-card-image.

The bitstream file uploaded successfully, and the LED light cycles on and off. But the final result cannot establish network communication. Is there a image file available for me to download? Please let me know if possible.

Thank you again for the great work.