antmicro/rowhammer-tester #201: Flashing bitstream fails

pjattke, 2025-01-10:
Flashing a bitstream on our DDR5 tester (S/N: 23) fails when trying to write the bitstream.

**Steps to Reproduce:**

1. Build a bitstream, e.g., `ddr5_tester-main-dfabb627_dimm_325.zip`
2. Invoke `make flash` (same with `make upload`).

**Observed Behavior:**

The FTDI device cannot be opened:

```
❯ make upload
openFPGALoader --board antmicro_ddr5_tester --freq 3e6 build/ddr5_tester/gateware/antmicro_ddr5_tester.bit
unable to open ftdi device: -4 (usb_open() failed)
JTAG init failed with: unable to open ftdi device
make: *** [Makefile:92: upload] Error 1
```

**Expected Behavior:**

The bitstream is flashed on the FPGA.

**Additional details:**

We found that unlike on our other machines, this machine reports multiple FTDI devices when the FPGA is connected:

```
❯ ls -la /dev/ttyUSB*
crw-rw-rw- 188,0 root 10 Jan 11:24 /dev/ttyUSB0
crw-rw-rw- 188,1 root 10 Jan 11:24 /dev/ttyUSB1
crw-rw-rw- 188,2 root 10 Jan 11:24 /dev/ttyUSB2
crw-rw-rw- 188,3 root 10 Jan 11:24 /dev/ttyUSB3
```

More details of these devices:

```
❯ lsusb
Bus 009 Device 003: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 009 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0e8d:0616 MediaTek Inc. Wireless_Device
Bus 003 Device 002: ID 048d:5702 Integrated Technology Express, Inc. ITE Device
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mtdudek, 2025-01-10:
Hi Patrick,

Could you run  `groups $(whoami)`  on the user that has issues to load the bitstream?
From the provided logs I can't see any issues that could point to the project's SW or HW.

Regarding  multiple FTDI devices in the `ls /dev/ttyUSB*` showing 4 available interfaces.
DDR5 tester uses FTDI device with 4 channels. Before any configuration, each one is visible in 
the /dev/ as an ttyUSB interface. `lsusb` correctly reports a single FTDI device.

pjattke, 2025-01-10:
Hi @mtdudek,

I rechecked and the user is in the correct `dialout` group. Also adding the user to the `tty` group, as some posts in the web suggested, did not help.

However, the [solution](https://github.com/trabucayre/openFPGALoader/issues/245#issuecomment-1213510964) of adding a udev rule and rebooting, as proposed in https://github.com/trabucayre/openFPGALoader/issues/245, solved the issue. The dev file permissions now look like:

```
❯ eza -lg /dev/ttyUSB*
crw-rw-rw- 188,0 root dialout 10 Jan 17:31 /dev/ttyUSB0
crw-rw-rw- 188,1 root dialout 10 Jan 17:30 /dev/ttyUSB1
crw-rw-rw- 188,2 root dialout 10 Jan 17:30 /dev/ttyUSB2
crw-rw-rw- 188,3 root dialout 10 Jan 17:30 /dev/ttyUSB3
crw-rw---- 188,4 root dialout 10 Jan 17:30 /dev/ttyUSB4
```

Thank you very much!

closed 2025-01-10 by pjattke